gdoor-org/gdoor #1: How to build an Gira Bus <-> Smart Home adapter

jschroeter, 2023-08-02:
Hello, 
Thanks a lot for your efforts on documenting!

I’m looking for a (self-made) solution to open our house door via Home Assistant. I know there are commercial solutions like Gira TKS-IP-Gateway. But it’s too pricy.

I’m wondering if you could give me some guidance on how to build the hardware to connect to the bus? I’m a software guy, don’t know much about hardware. But e.g. I’m a bit familiar with ESPHome.

Thanks and have a nice day!

DaSchaef, 2023-08-02:
Hi :-)

I have two working solutions, but they are still comparable complicated and need at least some knowledge about electronics.

The easiest thing is just listening to the bus. This can be done with little hardware.
Controlling the bus is more complicated.

And the most complex situation is reading/writing bus and beeing supplied by it (as it is "hard" to not disturb it). 

I have a custom build PCB by JLCPCB which can do all of that, but  bus supply is working only "border line" according to my measurements.
I know what roocky mistake I made but never corrected it, this is why I did not publish it.

So there are two ways:
- ESP way, few parts, easy to build: Needs knowledge and probably an oscilloscope to fine tune the voltage levels (I never finished the "simple" design ).
- STM32 way, fully custom PCB working with few acceptable bugs (should not be powered by the bus itself, but 5V USB works well): Needs knowledge about JLCPCB, some soldering, STM32 programming (JTAG) ...

I would be happy to build a solution together with someone, document it for the public - but sadly there is some electronics knowledge required - as I fear it wouldn't work first try.

Is there anything where you would feel comfortable with? if you bring in the time, effort and willingness I can support it. But this is no easy solution. Sorry :-/

jschroeter, 2023-08-21:
Hi, thanks a lot for your quick answer and sorry for my late reply.

Before I continue, another question: is the protocol somehow encrypted? Or can I "simply" listen what command is sent when I open the door via the indoor device and re-send this command later again to open the door? Or how would I find out the dst?
I was reading regarding possible alternatives and found the [NUKI Opener](https://nuki.io/de/opener/) which sounds promising, but it's not officially compatible with Gira. Also [they claim Gira uses an encrypted bus](https://developer.nuki.io/t/opener-with-gira-ap-system-55/2914/2). With [some additional device from Gira it seems to be working](https://developer.nuki.io/t/nuki-opener-with-gira-ap-and-turkommunikations-tasterschnittstelle-2fach-121000-or-unterputz-schaltaktor-121100/4287).

DaSchaef, 2023-08-31:
Sry I forgot your message, was on vacation.

The bus is not encrypted at all. From an electrical standpoint it is just a bit complicated and Nuki is too simple for that. Cheap excuse from whoever 😅

The addresses can be sniffe with the same hardware d and values just copied to your own commands. You could even brute force it but the bus speed is a natural barrier, it would take weeks and why wait 😄

Of course you can just salvage an indoor station.
There are different revisions but I think the Nuki thread should give a good start.

DaSchaef, 2023-08-31:
Ah what you would be able to also do: replace the pic microcontroller on a original indoor station with an esp.

The existing microcontroller just receives and sends out PWM signals, each on a dedicated pin. The pwm is just the digital version of the signals I show in this repo. 

 But I have no turn-key solution so that would be maybe even more complicated. And more costly.

ravenst0ne, 2024-02-15:
Hey @DaSchaef , really nice work! 
May I ask which HW you used to generate the signals with an ESP32 and what's the difference compared to the STM32 solution? Also how you powered the controller using the power from the bus line would be interesting.

I have STM32 and ESP32 available and would like to give it a try after listening to the bus with an oscilloscope. 
Powering with external 3,3/5V is not a problem, better have cables in front of the front door than having to walk to the door each time :)

DaSchaef, 2024-02-16:
Hi :)

I never beautified my project because:
1. I finished my goal
2. Continuing would have meant re-spin PCBs (money)
3. Receiving money would mean an obligation for a project which I only play with part-time
(not good).

This is why I search people who are able to proceed where I finished, and I'm fully willing to support it, but time and money I don't want to finish it myself :)
I just uploaded my internal, ugly stm32 design.

1.
Bus powering: Do not do it, without really knowing electronics ;-)
If you really understand my shitty schematic, we can discuss it in more details.
But believe me, there are audio and EMC related issues.

2.
RX is most easiest:
You can use something like, AC coupling the signal to a GPIO.
But limit the voltage with low-forward-voltage diodes (e.g. Schottky),
to not damage the GPIO in the rare case that a large voltage transient is coupled
to the GPIO.
[image]

After that you can use EXT Interrupts and Timers to decode the signals.
Logic should be simple, I should be able to help you:
In principle:
Use EXT INT to count the pulses inside a pulse train,
Use a TIMER/ALARM to see if a pulse train is finished (xx ms no EXT INT means pulses are over).
Add the pulse counts to an array.

If your timer fired and there are no counts (which means the timer fired twice in a row), you know that the whole transmission is over.
Think of the timer similar to a watchdog, where the EXT INT resets it.

Now you have an array with pulse counts, see the image on the GDOOR main repo.
You can then process this further to bits (>50 counts = start bit, >25 counts = Bitvalue is 0, <20 counts = Bitvalue is 1), and then you have a 01010010 bitstream, which you can
pack into 9 bit words, check or ignore parity bit, remove it, and now you have 8 bit value (aka chars).

3. TX is complicated,
I successfully breadboarded two possibilities:
3.1 First options was a Wien-Oscillator which I was able to turn on and off by a GPIO.
3.2. I think more easier is a bandpass: Just output the needed signals as a PWM, the bandpass will "make it a somehow nice sine trains", like shown in the top README.md.
A bandpass example with added output buffer is shown in the stm32-ugly.

4. Uploades STM32 design:
You should be able to procure the hardware pre-assembled from JLCPCB,
all assembly info are inside the schematic.
You need KiCAD with the "Fabrication Toolkit" PlugIn, this combo will create you production files, which you can upload to the JLCPCB webpage and they will produce, assemble and ship the PCBs.

At least, that is what I did a year ago.
Be aware of the two bugs I documented! They are still included!

1. Crystal Osc. Pinout is wrong, this should be corrected, otherwise you have a hard time
to manually correct this error (which means fine soldering needed!).
From my memory I swapped the meaning of the pins.

2. Remove R15, best is to set it to DNP (Do not populate) for JLCPCB,
this will disable self-powering from the gira bus, but leaves you with the options to power via the USB-C connector. (The whole board only needs ~ 10mA, at 5V this would be 0.05W)

Be also aware of the strange form factor: You can break-out the two circled PCBs, so that
the finished assembly fits into an "Unterputzdose" (German).

DaSchaef, 2024-02-16:
On thing I forgot:
The STM32 version is rock solid in RX and TX, only with the bugs (which I fixed manually) and the "strange" form factor.

DaSchaef, 2024-02-23:
I just uploaded a ESP32 version.

It can be produced by JLCPCB, 5 parts costs approx. 30€ (shipping and taxing increases price).
I currently have no firmware for this. If we have enough interest, I can order it.
(Everyone can order these as well, you can use the production subfolder in esp32).

Additionally needed are:
- ESP32mini 
e.g. https://www.az-delivery.de/en/products/esp32-d1-mini
- Pin Headers to connect/solder the ESP32 to the PCB
(if JLCPCB should solder pin headers it increases price by I would assume 20-30€
just for these).
- Firmware

If someone else (or I) order a batch, I can offer to write a simple firmware for it.
No full guarantee that there is no hardware bug in the design, but as I copied
it from the working STM32 PCB, risk is low.

[image]

[image]

ravenst0ne, 2024-02-23:
Unfortunately I didn't have the time yet to check your latest commits, but they look quite very interesting!

I'm totally interested in ordering such sample, feel free to contact me via mail for further details :)

jschroeter, 2024-02-23:
Same, I'm also interested. Let's sync via mail.

DaSchaef, 2024-02-25:
I created a private repo, due to missing mails and no DM capability on Github :-)

closed 2024-04-02 by DaSchaef