gdoor-org/gdoor #5: Add high-level props to JSON output

jschroeter, 2024-03-29:
To allow comfortable automations/triggers in Smart Home platforms like Home Assistant.

Suggestion for new fields:
- action: doorBell / floorBell / openDoor / ...
- param: which button was pressed
- deviceType: doorStation / indoorStation / controller
- source
- destination

DaSchaef, 2024-03-30:
We can re-use (and migrate to class based C++) from the STM32 code:
https://github.com/gdoor-org/gdoor/blob/a9e28f83e366b4aa6abce5f1029c4be776966766/firmware/stm32-ugly/Core/Inc/gdoor_protocol.h (and corresponding C file)

DaSchaef, 2024-04-01:
Can you test #6 ?
I expect that it will **not** work on first try, as it is a dry run implementation,
with significant refactoring in the RX logic.

Additionally this should also fix #4 without the big ArduinoJSON library dependency.

DaSchaef, 2024-04-03:
Closed in #6

closed 2024-04-03 by DaSchaef