gdoor-org/gdoor #8: Last action gets repeated?

jschroeter, 2024-04-06:
I think there might be a bug in the firmware: apparently the last received action is repeated every now an then. In the screenshot there is a log of the actions received by Gdoor (note the increasing event_id). E.g. there are actions received during night time where there was clearly no user interaction. Maybe there are (yet unknown) things going on on the bus and due to a bug the last received action is detected again?
Unfortunately I don't have a debug mode log recorded yet - I'll try to record one asap.

[image]

DaSchaef, 2024-04-06:
Reading the code,
there may be a bug in https://github.com/gdoor-org/gdoor/blob/main/firmware/esp32/gdoor/src/gdoor_data.cpp#L56

I'll try to commit a easy fast fix

DaSchaef, 2024-04-06:
Please check :)

jschroeter, 2024-04-07:
Looks good, thanks!
Do you know what triggered these?

commit 23e5160c (2024-04-06): Try to fix #8
closed 2024-04-07 by jschroeter