gdoor-org/gdoor #9: Add video and other bus actions to docs

jschroeter, 2024-04-08:
Collecting some more bus actions to add them to the docs, code + streamline wording. Work in progress.

## Outdoor ring & accept audio call & open door & manual close call
```
{"action": "BUTTON_RING", "parameters": "0360", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "011011A286FD0360A04A", "event_id": "12"}
{"busdata": "011011A286FD0360A04A", "data": ["0x1", "0x10", "0x11", "0xA2", "0x86", "0xFD", "0x3", "0x60", "0xA0", "0x4A"], "valid": true}

# indoor station automatically tells the outdoor station to start the video stream
{"action": "CALL_ACCEPT_VIDEO", "parameters": "0278", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "022028ED3F4B0278A1A286FD01", "event_id": "13"}
{"busdata": "022028ED3F4B0278A1A286FD01", "data": ["0x2", "0x20", "0x28", "0xED", "0x3F", "0x4B", "0x2", "0x78", "0xA1", "0xA2", "0x86", "0xFD", "0x1"], "valid": true}

# user activates 2-way audio by pressing the "phone" button on the indoor station
{"action": "CALL_ACCEPT", "parameters": "0000", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "020021ED3F4B0000A1A286FD60", "event_id": "14"}
{"busdata": "020021ED3F4B0000A1A286FD60", "data": ["0x2", "0x0", "0x21", "0xED", "0x3F", "0x4B", "0x0", "0x0", "0xA1", "0xA2", "0x86", "0xFD", "0x60"], "valid": true}

# user opens the door by pressing the "lock" button
{"action": "OPEN_DOOR", "parameters": "0060", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "020031ED3F4B0060A1A286FDD0", "event_id": "15"}
{"busdata": "020031ED3F4B0060A1A286FDD0", "data": ["0x2", "0x0", "0x31", "0xED", "0x3F", "0x4B", "0x0", "0x60", "0xA1", "0xA2", "0x86", "0xFD", "0xD0"], "valid": true}

# user closes the audio & video call by pressing the "phone" button again
{"action": "CALL_CLOSE", "parameters": "0000", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "020020ED3F4B0000A1A286FD5F", "event_id": "16"}
{"busdata": "020020ED3F4B0000A1A286FD5F", "data": ["0x2", "0x0", "0x20", "0xED", "0x3F", "0x4B", "0x0", "0x0", "0xA1", "0xA2", "0x86", "0xFD", "0x5F"], "valid": true}

# unclear what this is doing
{"action": "ACTION_UNKOWN", "parameters": "0000", "source": "000000", "destination": "000000", "type": "TYPE_UNKOWN", "busdata": "E20020ED3F4B0000A1A286FD5F", "event_id": "17"}
{"busdata": "E20020ED3F4B0000A1A286FD5F", "data": ["0xE2", "0x0", "0x20", "0xED", "0x3F", "0x4B", "0x0", "0x0", "0xA1", "0xA2", "0x86", "0xFD", "0x5F"], "valid": false}
```

## Outdoor ring + open door (without accepting audio call)
```
{"action": "BUTTON_RING", "parameters": "0360", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "011011A286FD0360A04A", "event_id": "29"}
{"busdata": "011011A286FD0360A04A", "data": ["0x1", "0x10", "0x11", "0xA2", "0x86", "0xFD", "0x3", "0x60", "0xA0", "0x4A"], "valid": true}
{"action": "CALL_ACCEPT_VIDEO", "parameters": "0278", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "022028ED3F4B0278A1A286FD01", "event_id": "30"}
{"busdata": "022028ED3F4B0278A1A286FD01", "data": ["0x2", "0x20", "0x28", "0xED", "0x3F", "0x4B", "0x2", "0x78", "0xA1", "0xA2", "0x86", "0xFD", "0x1"], "valid": true}
{"action": "OPEN_DOOR", "parameters": "0060", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "020031ED3F4B0060A1A286FDD0", "event_id": "31"}
{"busdata": "020031ED3F4B0060A1A286FDD0", "data": ["0x2", "0x0", "0x31", "0xED", "0x3F", "0x4B", "0x0", "0x60", "0xA1", "0xA2", "0x86", "0xFD", "0xD0"], "valid": true}
{"action": "CALL_CLOSE", "parameters": "0000", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "020020ED3F4B0000A1A286FD5F", "event_id": "32"}
{"busdata": "020020ED3F4B0000A1A286FD5F", "data": ["0x2", "0x0", "0x20", "0xED", "0x3F", "0x4B", "0x0", "0x0", "0xA1", "0xA2", "0x86", "0xFD", "0x5F"], "valid": true}
{"action": "CALL_CLOSE", "parameters": "0000", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "013020A286FD0000A016", "event_id": "33"}
{"busdata": "013020A286FD0000A016", "data": ["0x1", "0x30", "0x20", "0xA2", "0x86", "0xFD", "0x0", "0x0", "0xA0", "0x16"], "valid": true}
```

## Button ring + no user interaction
```
{"action": "BUTTON_RING", "parameters": "0360", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "011011A286FD0360A04A", "event_id": "24"}
{"busdata": "011011A286FD0360A04A", "data": ["0x1", "0x10", "0x11", "0xA2", "0x86", "0xFD", "0x3", "0x60", "0xA0", "0x4A"], "valid": true}
{"action": "CALL_ACCEPT_VIDEO", "parameters": "0278", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "022028ED3F4B0278A1A286FD01", "event_id": "25"}
{"busdata": "022028ED3F4B0278A1A286FD01", "data": ["0x2", "0x20", "0x28", "0xED", "0x3F", "0x4B", "0x2", "0x78", "0xA1", "0xA2", "0x86", "0xFD", "0x1"], "valid": true}

# indoor station closes the call after ~30s (time seems to be configurable in the indoor station)
{"action": "CALL_CLOSE", "parameters": "0000", "source": "ED3F4B", "destination": "A286FD", "type": "INDOOR", "busdata": "020020ED3F4B0000A1A286FD5F", "event_id": "26"}
{"busdata": "020020ED3F4B0000A1A286FD5F", "data": ["0x2", "0x0", "0x20", "0xED", "0x3F", "0x4B", "0x0", "0x0", "0xA1", "0xA2", "0x86", "0xFD", "0x5F"], "valid": true}

# also the outdoor station seems to close the call after a timeout
{"action": "CALL_CLOSE", "parameters": "0000", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "013020A286FD0000A016", "event_id": "27"}
{"busdata": "013020A286FD0000A016", "data": ["0x1", "0x30", "0x20", "0xA2", "0x86", "0xFD", "0x0", "0x0", "0xA0", "0x16"], "valid": true}
```

## View missed call / pressing the "menu" button on the indoor station
When viewing the missed call on the indoor station ("phone" button blinking + pressing it), the following command appears on the bus, although the video stream doesn't start or at least is not visible on the indoor station. It's just showing the picture which is saved on the SD card of the indoor station.
The same happens when pressing the "menu" button on the indoor station (there is no active call, no video showing, just the menu is visible).
```
{"action": "CALL_ACCEPT_VIDEO", "parameters": "0100", "source": "ED3F4B", "destination": "000000", "type": "INDOOR", "busdata": "013028ED3F4B0100A172", "event_id": "28"}
{"busdata": "013028ED3F4B0100A172", "data": ["0x1", "0x30", "0x28", "0xED", "0x3F", "0x4B", "0x1", "0x0", "0xA1", "0x72"], "valid": true}
```

## Open door without open call
Similar to opening with an active call, but the destination is the controller instead of the outdoor station.
```
{"action": "OPEN_DOOR", "parameters": "0060", "source": "ED3F4B", "destination": "A25479", "type": "INDOOR", "busdata": "020031ED3F4B0060A1A254791A", "event_id": "11"}
[... 48 lines cut ...]

# 3
{... "parameters": "0360" }

# 4
{... "parameters": "0460" }

# 5
{... "parameters": "0560" }

# 6
{... "parameters": "0660" }

# 7
{... "parameters": "0760" }

# 8
{... "parameters": "0860" }

# 9
{... "parameters": "0960" }

# 10
{... "parameters": "0A60" }
```

jschroeter, 2024-04-08:
Ideas:

- rename `CALL_ACCEPT_VIDEO` to `VIDEO_REQUEST` or similar, since a) it's not a call (video is 1-way) and b) the indoor station typically sends this action automatically without any user interaction. The audio call is activated as a separate action.
- rename `BUTTON_RING_UNKOWN` to `SWITCH`?
- add action `CALL_INTERNAL` for `0x12`
- rename `OPEN_DOOR` to `DOOR_OPEN`, just to have consistent naming?

jschroeter, 2024-04-08:
@ravenst0ne if you have a few minutes, please have a look. Maybe your station even supports more?

DaSchaef, 2024-04-08:
> Ideas:
> 
>     * rename `CALL_ACCEPT_VIDEO` to `VIDEO_REQUEST` or similar, since a) it's not a call (video is 1-way) and b) the indoor station typically sends this action automatically without any user interaction. The audio call is activated as a separate action. As mentioned above, the only weird thing is that this action also is sent when just opening a missed call or opening the menu of the indoor station.

Yes, we should then also use `AUDIO_REQUEST`.
Maybe AUDIO_REQUEST is the one which is also triggering video and VIDEO_REQUEST is something totally different.
We can only guess :)

> 
>     * rename `BUTTON_RING_UNKOWN` to `SWITCH`?

This is funny :-) My audio station (indoor and outdoor) sends `BUTTON_RING_UNKOWN` for all *unassigned* buttons.
The generic "light" button on my indoor station sends 0x41. So maybe change 0x41 to LIGHT and 0x42 to BUTTON
("Schalthandlung", is it really a switch or a "press action"?). But maybe also the "Schaltform" (button vs. swicth) defines if it is 0x41 or 0x42.
We only can guess at the moment.

> 
>     * add action `CALL_INTERNAL` for `0x12`
> 
>     * rename `OPEN_DOOR` to `DOOR_OPEN`, just to have consistent naming?

Sounds good!

DaSchaef, 2024-04-08:
> # unclear what this is doing
> {"action": "ACTION_UNKOWN", "parameters": "0000", "source": "000000", "destination": "000000", "type": "TYPE_UNKOWN", "busdata": "E20020ED3F4B0000A1A286FD5F", [...]

`valid = false` Seems an invalid message (checksum and/or parity bit).
Does this happen often? This may indicate a "weak" message which we do not receive properly.
In contrast, the "ACTION_UNKOWN" even though the field is 0x20 may also indicate that some pointers were dangling,
maybe because to message were shortly after each other on the bus or so.
So it could be S/W or H/W or just a bus glitch.

I also noticed a bug in the debug output, the raw counting values (which would help in such a case) are not send.
Doh!  I'll fix that.

DaSchaef, 2024-04-08:
`E20020ED3F4B0000A1A286FD5F`
you also received
`020020ED3F4B0000A1A286FD5F`

It seems the first nibble is mangled.

DaSchaef, 2024-04-10:
Hm, the debug output indicates a valid bus transmission with a wrong bit.
The bit length for the right commands (excerpt) were:
`D | 21 | D | 21 | 21`
for the invalid message
`D | 21 | D | B | 20`

So one bit was inverted (the 0x0B).
All bits around are valid, so it was also not a single glitch/transient
and 0x0D vs. 0x0B - this is also no far.

Bit = 0 = 0x21 = 33 periods wide pulsetrain
Bit = 1 = 0xD =  13 periods wide pulsetrain
Bit = ERROR = 0xB =11 periods wide pulsetrain

At the moment, nothing we can do I would assume,
but let's keep an eye on it.

jschroeter, 2024-04-10:
Alright - since invalid messages don't get printed in the non-debug output, I also don't see any issue with it.

> The generic "light" button on my indoor station sends 0x41. So maybe change 0x41 to LIGHT and 0x42 to BUTTON
("Schalthandlung", is it really a switch or a "press action"?)

Same here, the "light" button on my indoor station sends 0x41, so I would leave it as is. Unfortunately I couldn't test the light button on the outdoor station yet, it seems to be not connected (I will check one day).
On my indoor station the "Schalthandlung" is only accessible via the menu, so there is no separate button for it. And in the menu, for each of the 4 there are two options: on and off, which result in different parameters being sent (on: `xx40`, off: `xx50`), that's why I thought about "SWITCH". But I guess it's better to keep it generic for now and call the action "BUTTON" (instead of "BUTTON_RING_UNKOWN").

jschroeter, 2024-04-10:
I'm not 100% happy with "CALL_CLOSE" as this is also used for ending the video stream, meaning there was no active "call". But I don't have a better idea yet, so I left it as is.

DaSchaef, 2024-04-10:
LGTM :+1:

jschroeter, 2024-04-11:
Idea: `CALL_CLOSE` -> `AUDIO_VIDEO_END`

jschroeter, 2024-04-12:
I updated the docs, please have a look and then we can close this issue.

closed 2024-04-15 by DaSchaef