gdoor-org/gdoor #12: Build firmware via GitHub Actions & publish releases

jschroeter, 2024-04-17:
To be able to download the firmware & upload it via Wifi web UI.

DaSchaef, 2024-04-17:
:+1: 
I think we need an extra repo for that, + github.io page :)

link collection:
https://esphome.github.io/esp-web-tools/
https://github.com/witnessmenow/ESP-Web-Tools-Tutorial

https://docs.platformio.org/en/stable/integration/ci/github-actions.html
https://gist.github.com/marcovannoord/4275a25639427e13d6e74f17636bbd5f
https://docs.github.com/de/actions/using-workflows/events-that-trigger-workflows#running-your-workflow-only-when-a-push-of-specific-tags-occurs

jschroeter, 2024-04-18:
I think this could also be within one repo + a gh-pages branch for the deployed website. I personally would prefer that, I think it's easier to maintain. But up to you.

DaSchaef, 2024-04-19:
I do not like branches that much, as they are "invisible" on filesystem level?
Also I'm not familiar with Github Actions at all, in contrast to Jekyll ;-)

So I kind of feel "safer" to play in another repo.
(Another benefit is the github URL, where repo name needs to match org. name)
https://gdoor-org.github.io/
https://github.com/gdoor-org/gdoor-org.github.io

(I have no clue how the automatic compiling is set up)

jschroeter, 2024-04-22:
> I do not like branches that much, as they are "invisible" on filesystem level?

True, but I was wrong, it also works without a `gh-pages` branch: I gave it a try in
https://github.com/jschroeter/gdoor/ and added the website to the /website folder.

> Also I'm not familiar with Github Actions at all, in contrast to Jekyll ;-)

Both work very well together, just add this file and it will take the /website folder, build it with Jekyll and publishes the result to GitHub Pages: https://github.com/jschroeter/gdoor/blob/main/.github/workflows/website.yml

https://github.com/jschroeter/gdoor/actions/runs/8789883341
https://jschroeter.github.io/gdoor/ (path to CSS is broken but would be easy to fix)

> Another benefit is the github URL, where repo name needs to match org. name

True, this seems to be a good argument to keep it separate. And there is no issue with it, I just thought if everything is in one repo we might be able to re-use the GDoor logo and other content to not maintain it multiple times.

jschroeter, 2024-04-22:
Regarding the firmware build, here a first PR: https://github.com/gdoor-org/gdoor/pull/13

What should follow in next PRs:
- combine the firmware files and add a manifest compatible with esp-web-tools
- create releases on GitHub
  - we could do this manually
  - or better automatically, e.g. with https://github.com/softprops/action-gh-release. I've not tried it yet but this action sounds promising.

jschroeter, 2024-04-26:
Good news, I was able to successfully flash the adapter via esp-web-tools! Cool stuff, also the integrated terminal is quite helpful. In case you want to try it: https://jschroeter.github.io/gdoor/web-installer.html
[image]

But we need to decide: Where should we place the manifest.json and firmware file?

I thought the best would be to upload both in the release process (meaning whenever we create a release, e.g. with the GitHub release action mentioned above it would happen automatically when adding a tag) as release assets. Since the manifest.json also can include a version number, it makes sense that it gets dynamically generated during the release. On the website we could [directly link the file of the latest release](https://docs.github.com/en/repositories/releasing-projects-on-github/linking-to-releases), e.g. `<esp-web-install-button manifest="https://github.com/gdoor-org/gdoor/releases/latest/download/manifest.json">`. The generated manifest could relatively point to `firmware-merged.bin`, this should work according to the docs.
But [I tried it](https://github.com/jschroeter/gdoor/releases) and unfortunately GitHub doesn't seem to set the according CORS headers, meaning the browser blocks directly downloading the manifest/firmware file from the release asset. There is a [discussion](https://github.com/orgs/community/discussions/45446) for it.

So I do see 2 options:

1. push the manifest and firmware to the gdoor or website repository

The manifest doesn't change often, except of the version number, but probably it's not so important as you only need it for the initial installation. But the pushed static firmware would become outdated quickly.

2. add a little service (e.g. [Netlify function](https://docs.netlify.com/functions/overview/)) which proxies the request to the release assets and adds correct CORS headers

Would work, is probably free for our use case and only requires a few lines of code. But introduces a new platform which I would like to avoid.

DaSchaef, 2024-04-29:
Sounds great! Can we make an action to push it to the homepage repo as "latest"?
2) Sounds the most simple for me?

DaSchaef, 2024-05-04:
Done, I followed the instructions in the link.

jschroeter, 2024-06-21:
Some progress on this: firmware changes on the main branch now automatically get pushed to the website branch, see https://github.com/gdoor-org/gdoor/blob/main/.github/workflows/firmware.yml#L45 and e.g. https://github.com/gdoor-org/gdoor-org.github.io/commit/6ff8995dd7883af7ae5232ad17a2a620e78eb075.

Also I tried setting the GDOOR_VERSION env variable, but it's not working yet, pretty sure the issue is in https://github.com/gdoor-org/gdoor/blob/main/.github/workflows/firmware.yml#L32, probably needs some other escaping.

If all this works we can adjust it to only push on release tags, meaning not everytime the main branch changes. How do you think we should handle hardware and firmware releases? Does every hardware release also include a firmware release? But firmware releases don't neccessarily include a hardware release?

DaSchaef, 2024-06-22:
I would say separate firmware / hw releases. 
Great work!

jschroeter, 2024-06-22:
Makes sense. How should we differentiate them in the Git tags (so I can setup the release GitHub action)?
Should we pre/suffix the Git tags? E.g. v3.1-hardware and if it’s not „-hardware“, it’s a firmware. Because for sure there will be more firmware then hardware releases.

DaSchaef, 2024-06-22:
Good idea!

jschroeter, 2024-06-22:
Alright, I think this can be closed soon:

When pushing a tag starting with `v` and not ending with `-hardware`,
- a draft release gets created including the firmware assets.
- it's draft, so e.g. emails to repo subscribers only go out when we added the release notes and published the release via the GitHub UI (sorry for the spam, I should have enabled this right from the start)
- also the web installer firmware gets pushed to the website repo.

Open issue:
- the GDOOR_VERSION environment variable is set, so the `manifest.json` and also the firmware should include the version number (e.g. visible in Home Assistant device details). It's in the format `v1.0.1+hash` to be valid semver. But somehow it doesn't end up in HA yet, there it still says `dev`. But [prepare-web-installer.py](https://github.com/gdoor-org/gdoor/blob/main/firmware/esp32/gdoor/prepare-web-installer.py#L42) is able to receive it, so it gets added to the manifest.json. Do you have any idea why it's not picked up in the firmware yet?

After we solved this and as a verification: Could you rename the existing tags to the new format and create one for the latest firmware? Thanks.

DaSchaef, 2024-06-23:
For platform IO you need to set another env. variable
(https://community.platformio.org/t/pass-global-define-at-run-time/1439/8)

You need to differentiate between bash level environmental variable
and gcc `#define` which is set via gcc `-D` command line option,
which you can extend via the env. variable `PLATFORMIO_BUILD_FLAGS`

`MUH5678964` as example version value, i think you can put in your $GDOOR_VERSION here:

1. `export export PLATFORMIO_BUILD_FLAGS="-DGDOOR_VERSION=\\\"MUH5678964\\\""`
(Note the escaping, otherwise the `"` get lost on gcc level, may be different in github jobs)
2. `pio run`
3. Check that version was set, e.g. `strings .pio/build/GDOOR_ESP32MINI/web/firmware_merged.bin | grep MUH` should return `MUH5678964`

jschroeter, 2024-06-23:
Thanks, it works now! Please give it a try :-)

DaSchaef, 2024-10-10:
The release process seems to start also on other branches?
Could you check?
https://github.com/gdoor-org/gdoor/actions

DaSchaef, 2024-10-10:
Ah I see, release is only with tag. Build is OK. Sry for the noise.

closed 2024-10-10 by DaSchaef