gdoor-org/gdoor #15: Merge debug output into single JSON

jschroeter, 2024-04-27:
Not urgent but I think we should invest the effort to only send 1 JSON per event. Initially I thought only Home Assistant gets confused when sending multiple, but it seems that MQTT Explorer also doesn't recognize the JSON format if Gdoor debug mode is enabled:
[image]

DaSchaef, 2024-04-29:
What if we send debug to <mqtt-tx-topix>/debug ?
Should be most easiest solution to implement :smile:

jschroeter, 2024-05-02:
To not confuse Home Assistant, MQTT Explorer and possibly other tools, we should:

- have a consistent output format: I would suggest we stick to valid JSON in all cases, meaning also the debug output could look like `{"debug": "Received data from bus"}`
- make sure we always send only one message (not two JSONs in one MQTT event or missing new lines for serial)

For MQTT, I do like your idea of moving all debug output into a separate topic. Easy to implement + less conflicts, e.g. no risk of breaking smart home automations while debug mode is on.

For serial, I would say we mix normal and debug output like we do now. As long as we take care about the two points above, we're good. The only thing I'm wondering is if we should merge both JSONs like I initially suggested above, it seems weird to send two partly similar JSONs for one and the same bus event. So we could also do the following: if debug mode is on, the `raw` field is added to the regular bus event JSON (for MQTT and serial). So only the debug messages like "GDOOR Setup start", "Received data from bus" etc. are sent to the MQTT debug topic. Does this make sense?

DaSchaef, 2024-05-04:
I would not send debug output messages via MQTT (as it should be with no Wifi dependency which may break in a debug case).
Rest is implemented in #16

PR #16 (2024-05-05): Better output format for home automation
Fixes #15
closed 2024-05-05 by jschroeter