gdoor-org/gdoor #17: MQTT Discovery

DaSchaef, 2024-05-08:
I'm collecting infos about MQTT  Discovery

```
{
    "sw_version": "3.0",
    "name": "GDOOR Gira Bus Adapter",
    "force_update": true,
    "icon": "mdi:door",
    "state_topic": "gdoor/rx",
    "command_topic": "gdoor/tx"
}
```

jschroeter, 2024-05-09:
Just for reference:

- https://www.home-assistant.io/integrations/mqtt#mqtt-discovery
- https://www.openhab.org/addons/bindings/mqtt.homeassistant/

Nice to have: set the `configuration_url` to the IP of the adapter to get a link to the adapter UI (see https://developers.home-assistant.io/blog/2021/10/26/config-entity/).

DaSchaef, 2024-05-09:
`entity_category=config`:  we can also create config entries.
   `entity_category=diagnostic` we can show status infos

jschroeter, 2024-05-09:
Maybe useful as inspiration: https://github.com/tbnobody/OpenDTU/blob/main/src/MqttHandleHass.cpp

DaSchaef, 2024-05-10:
```
{
    "name": "Bus Data",
    "force_update": True,
    "icon": "mdi:door",
    "state_topic": "gdoor/rx",
    "command_topic": "gdoor/tx",
    "json_attributes_topic": "gdoor/rx",
    "value_template": "{{ value_json.action }}",
    "uniq_id": "wfffedd_data",
    "device": {
        "sw_version": "3.0",
        "name": "GDOOR Adapter",
        "model": "ESP32 (de:ad:be:ef)",
        "manufacturer": "GDOOR Project",
        "configuration_url": "http://127.0.0.1",
        "ids": "hdisep"
    }
}
```

DaSchaef, 2024-06-16:
Closing with #18 
Future work started in #25

closed 2024-06-16 by DaSchaef