gdoor-org/gdoor #19: Spare bus adapters

nholloh, 2024-05-16:
Does anyone, preferably from Germany, have a spare bus adapter lying around?

If not, I may proceed with an order. If you're looking to source one through me, feel free to leave a message here.

DaSchaef, 2024-05-16:
Germany is no problem. I do not need an adapter, but I can support if needed.

Please note:

1. Currently the V3.1 (latest version) is not in a release file or production folder.
You need to create the JLCPCB data from KiCAD (there is also a JLCPCB plugin which I use).
If needed, I can create the production files.
2. The version 3.0 had some bugs, which we hope to have fixed in 3.1, but we did not produce 3.1 yet.
We changed:
- Way smaller board, more dense, more tiny parts.
- We changed the bus connector from a screwing type to a plugin-type.
JLCPCB should ship a matching cable end together with the boards, but we have not yet tested if this is really the case :)
- The powering scheme is changed a bit
- We implemented a fix for receiving the data, via comparator
- We increased the sending amplification.

So all in all, I think it should work, but as its the first run for 3.1, there is a risk that we may find other bugs.
I can assure that I can try to support it - we got all v3.0 boards working :)

jschroeter, 2024-05-16:
I‘m interested in getting a V3.1 adapter and then would have a working V3 one for sale in case someone is interested.

DaSchaef, 2024-05-16:
If there is interest, I can offer to do the JLCPCB procurement again - to test the PCBs here on my test setup.
If you share the cost, and I can keep one for development? (same as 3.0)

nholloh, 2024-05-16:
@jschroeter I'd be up for 3.1 as well. @DaSchaef sounds good to me!

DaSchaef, 2024-05-16:
I created a private repo, so that we can better exchange non-public data.

mrtnkhl, 2024-05-17:
@DaSchaef I would be interested in a 3.1 unit as well, my Gira setup is a bit more complex and could be interesting for testing in a larger install.

sim-san, 2024-05-30:
@DaSchaef 
I found this just this morning. I have a gira door communication and and home assistant server. I would like to help you testing the board.

DaSchaef, 2024-05-31:
I'll contact you. We have one last adapter available.

github-k8n, 2024-06-19:
Anyone had any luck ordering from JLCPCB? they seem to currently be out of stock on the connector .. if anyone has done or wants to do another run ordering these, I'd be in..

DaSchaef, 2024-06-19:
See #29 :-)

DaSchaef, 2024-10-09:
Could you please look into #36 and give me feedback about stability (details are in #36) - the more people test, the better.
This would help me to free up some I/Os for audio support. No firmware update needed, only a setting in the web interface of GDoor.

ix2, 2024-12-17:
Hi, I am interested in buying a board. If @dtill  have one available I am really interested. BTW, my location ist Germany (Ruhrgebiet). How to handle it?

Shoomaan, 2024-12-17:
I‘ve got 4 Boards ( 3.1 )to sell. 
Shipping from Germany
12€ / each  + Shipping

dtill, 2024-12-17:
@Shoomaan  i'll take one as well :) let me know how!

> I‘ve got 4 Boards ( 3.1 )to sell. Shipping from Germany 12€ / each + Shipping

ix2, 2024-12-18:
@Shoomaan I have just send you a message on your 'Kleinanzeigen' ad. Would like to take two of the available boards.

Mr-Taxman, 2025-01-18:
> I have a four 3.1-1 available. Let me know if you're interested.

Would like to buy one, but how? Any standard procedure paypal friends?

Thanks Christof

ATell-SoundTheory, 2025-01-18:
> I have a four 3.1-1 available. Let me know if you're interested.

For those interested, please send an email to atell at soundtheory.com. Shipping is from Germany. Preferred payment is PayPal and I can also provide an ESP32mini with the GDoor PCB.

cbka, 2025-03-29:
> > I have a four 3.1-1 available. Let me know if you're interested.
> 
> For those interested, please send an email to atell at soundtheory.com. Shipping is from Germany. Preferred payment is PayPal and I can also provide an ESP32mini with the GDoor PCB.

i have 3 left (Kit with ESP32)

SoleroTG, 2025-05-04:
@ATell-SoundTheory, @cbka, or does anyone else have a spare kit for shipment in germany?

Luckymoep, 2025-05-04:
Hi,
does anyone have a spare to sell? I am located in Germany...
@cbka do you have one left?

dtill, 2025-05-04:
> Hi, does anyone have a spare to sell? I am located in Germany... [@cbka](https://github.com/cbka) do you have one left?

yes, have a look on Kleinanzeigen for "gdoor"  ;)

cbka, 2025-05-04:
Message ID: ***@***.***>Yes :-) 2 übrig

maxliem9, 2025-06-04:
@cbka I am also interested. Also located in Germany.
You may send me an mail? max.liem.public@gmail.com

debergi, 2025-06-29:
Does anyone have another one laying around that I could buy? Maybe someone from Germany

maxliem9, 2025-06-30:
@debergi , yes. I have 4 sets with the ESP32 mini.
I am from Germany. Send me your email and I will contact you with more info.

maxliem9, 2025-06-30:
Or just me an Email to max.liem.90@gmail.com

dagroe, 2025-08-21:
Does anyone still have one to spare or is interested in a pooled JLCPCB order?

dennisschoeni, 2026-08-19:
@struhs hello, i just wrote an email to you, im intrested in purchasing an adapter.

mjoebar36, 2026-09-13:
Hi there, I'm looking for a 3.1 adapter (with or without ESP), does anybody has a spare one to sell to me? @struhs ?
I tried to order adapter from JLCPCB, but the 3.1.1. assembly does give a few remarks (!) and I don't dare to choose from the (position) options. I live in The Netherlands and I try to integrate my Gira 1288 in the office in Home Assistant.
Thanks in advance.

struhs, 2026-09-14:
> Hi there, I'm looking for a 3.1 adapter (with or without ESP), does anybody has a spare one to sell to me? [@struhs](https://github.com/struhs) ? I tried to order adapter from JLCPCB, but the 3.1.1. [...]

Sorry, at least I am currently out-of-stock

mjoebar36, 2026-09-14:
> Sorry, at least I am currently out-of-stock

Thanks for responding Sir, I think I try to order via JLCPCB.

Shoomaan, 2026-09-14:
> Thanks for responding Sir, I think I try to order via JLCPCB.

I could sell you mine, we don't have a girl doorbell anymore.

1RandomDev, 2026-09-24:
If anyone is still looking for a spare one, I'm currently selling 4 pcs for 25€ each at https://www.ebay.de/itm/158333332227

mjoebar36, 2026-10-01:
> I could sell you mine, we don't have a girl doorbell anymore.

Thanks for the offer, I missed your reply and already ordered 5 pieces at JLCPCB. A bit disappointing experience I have to admit.
First they emailed me about the bus polarity and after that the connection was mounted as such that after soldering the mounting screws can't be reached anymore without removing the ESP. 
The price of around 50 euro for 5 pieces was calculated (PCB's and transport) and was OK, but later the Dutch customs wanted to have another 30 euro before they could be delivered ... :(. Appearantly a new law from the 1st of July '26, so be careful when you order from China.

Up to now I'm struggeling with getting the adapter to work correctly. It has been flashed with firmware and the GDoor bus is visible in Home Assistant via mqtt. 
Unfortunately it is switching between idle and unavailable when the doorbell is touched, looks like a protocol error. Switched the bus wires, changed the 5V power adapter with a stronger one (3A), but no result. Not sure what the problem is here.... any suggestions?

dtill, 2026-10-01:
Did u use the [ESPHome-component](https://github.com/gdoor-org/esphome-components) firmware and installed it via ESPhome Device Builder within Homeassistant? (That should work smooth and very stable)
or 
Did you install via the [gdoor-Web-Installer](https://gdoor-org.github.io/web-installer.html) referenced on the website? (the "gdoor-firmware" is good for testing and developing the hardware but for me not proven to work in long tern production especially because of not so stable wifi stack)

I personally recommand using ESPHome with this adapter. :)

mjoebar36, 2026-10-02:
> Did you install via the [gdoor-Web-Installer](https://gdoor-org.github.io/web-installer.html) referenced on the website? (the "gdoor-firmware" is good for testi

Hey, thanks for the tip, looks very promising. I will let you know if it works!

mjoebar36, 2026-10-03:
> Did u use the [ESPHome-component](https://github.com/gdoor-org/esphome-components) firmware and installed it via ESPhome Device Builder within Homeassistant? (That should work smooth and very [...]

Thanks again for the good work @dtill , I've got it working! Perfect!

NiXShankS, 2026-10-31:
Heyho guys,

i got a Gira ring aswell and wanted to ask you if someone got a spare adapter left for me? :)

Would be nice!
Thank You

PabloMangustin, 2026-11-02:
Hi there,

I am new to Gdoor. Does anyone have by any chance 2 extra bus adapters? I would be more than happy to buy them from you.

Thanks in advance.

Sebbstar, 2026-11-05:
Hi, I am searching for an adapter shipping to Germany.  Does anyone has surplus? :)