gdoor-org/gdoor #21: Video Support

mrtnkhl, 2024-05-17:
There is currently no video support planned until reverse engineering and interfacing with the digital bus is fully working. Creating this Issue as a placeholder for possible future work and to inform others who might be looking for video support.

Details concerning possible Audio support are tracked in Issue https://github.com/gdoor-org/gdoor/issues/20. 

Per @DaSchaef there is sufficient evidence to assume that video is transmitted in analog just like the audio part of the communication with the (door) station with a resolution of roughly 320xN pixels (or respective lines). It is assumed that a ADC would be needed to convert the analog signals to digital for the ESP32 and then send them via SIP/RTSP/RTP to a destination on the network. Issues outlined in https://github.com/gdoor-org/gdoor/issues/20 apply, in addition to those the extra bandwidth needed to digitize the video signal from the wire is going to be of concern.

mrtnkhl, 2024-05-17:
See this thread for a similar discussion for a Comelit system:
https://github.com/Elektroarzt/simplebus2-mqtt-bridge/issues/3

mrtnkhl, 2024-05-17:
https://www.eevblog.com/forum/projects/ic-for-capturing-composite-video-screenshot/

mrtnkhl, 2024-05-17:
https://github.com/wooky/esp32-analog-video-over-wifi

https://github.com/11tools/esp32cvbscam

jschroeter, 2024-05-17:
I can confirm that the resolution is 320 × 240 since the images saved on the SD card of the indoor station do have that resolution. They are saved as BMP files but I guess this is irrelevant.

jschroeter, 2024-05-17:
A bit off topic, but while I was setting up mobile notifications when someone hits our door bell I noticed that the Home Assistant mobile apps supports attaching a camera image/stream to the notification: https://companion.home-assistant.io/docs/notifications/notification-attachments#automatic-snapshots

I initially thought I don't need this, but of course, once I got the adapter running and read this, I immediately thought: that would be cool :-)

DaSchaef, 2024-05-23:
https://github.com/antmicro/cvbs-mipi-bridge
https://github.com/o-murphy/cvbs2rpi

DaSchaef, 2024-06-10:
I have a question to the owners of video installations:

Is the video parallel to the audio conversation?
If yes, the video signal may be modulated and no simple "cinch video", and we would need to find out the modulation scheme.

mrtnkhl, 2024-06-11:
Video and audio are transmitted in parallel. Not sure if audio is duplex but I think it is.

DaSchaef, 2024-06-11:
Hmm, we need someone with a video installation and at 40MHz bandwidth (better 100Mhz) oscilloscope.
We need to record start of video, to identify the video modulation/signal scheme.

I'm not 100% familar with analog video, but from what I understand: the standard CVBS goes down in the spectrum, overlapping with the audio range. So I would assume that the video may be FM (or even AM, but I don't think so) modulated to shift it to higher frequencies.

Or audio / video is interleaved? Don't know if such a thing could work

mrtnkhl, 2024-06-11:
Any recommendations for an affordable USB oscilloscope that would satisfy those requirements? My GIRA system has video.

DaSchaef, 2024-06-14:
I had a quick glance at Video Bus signals.

Sadly they seem to be modulated, over 100 MHz?
I had a bit of trouble with my cheap 100MHz oscilloscope.
It may be over 100Mhz or the cheap scope was crap :wink: 

I'll try a SDR solution next time, so that we first can get a feeling about the needed carrier frequency,
and be sure what expected range to measure and that we use the right measurement setup.

As the Modulation may be over 100 MHz it may indicate that standard analog TV modulation is used??
From Wikipedia:
[image]

DaSchaef, 2024-06-19:
Small status:
I'm still working on this, also on the hardware side of #20 
But as I want to combine both, it takes some time. I still need to find time for another round of measurement to get the modulation frequency.

DaSchaef, 2024-06-23:
Update (slow as I do not own a video bus):

I measured today via SDR (hackrf) and couldn't identify the modulation frequency.
This may indicate that the modulation frequency is on some VHF/UHF band where the RF signal  is hidden behind the normal over the air transmission like DVB-T etc.

So I either need to setup a wired coupler for the SDR (without destroying the sensitive SDR RF input) or I need to use the way more expensive oscilloscope (which I tried to avoid :wink: ).

So:

Easy option one (cheap oscilloscope): :-1: 
Easy option two (SDR with Antenna): :-1: 

meeh, its getting more complicated :sweat_smile:

mrtnkhl, 2024-06-23:
You got this! Curious to hear how they did it!

CENGOiSM, 2024-07-17:
I recently came across this project and wanted to express my gratitude for your work 🙏. While I’m not particularly knowledgeable about the electronic aspects, I found this FAQ that describes the frequencies used to transmit video:

https://partner.gira.de/service/faq/antwort.html?id=939

Center frequency **8 - 12 MHz**
Sidebands **+ / - 5 MHz**

Maybe this information might be helpful to you.

DaSchaef, 2024-07-17:
Wow, yes this is very helpful.

It indicates some problems during my oscilloscope trials.
Dual Sideband with +- are also a good thing. If it would be AM with both sideband this may ease demodulation significantly. No VSB like old time TV transmission reduces demodulation effort.

Let's see when I have access to a video system again. I'll test. Thanks!

DaSchaef, 2024-07-24:
[image]

We have first data, it should be a checker board :)

It is FM modulated, at  ~11MHz.
PAL, 625 , 25 FPS (these settings worked)

The demodulator in SDRangle is great but rather limited to analyze. This needs more custom software to better understand and also learn what is needed in hardware.
So it will still take time, no fast progress to be expected.

mrtnkhl, 2024-07-24:
GDoor's moon landing!

CENGOiSM, 2024-07-25:
Wow, great news! Would it be beneficial to share a 10-60 second segment of the transmitted signal so more people can analyze its integration potential through simulation?

DaSchaef, 2024-08-04:
A short status, I'm still - slowly in the free time - working on video.
Uploading is a bit hard:
- I have no good capture yet
- Raw data even for seconds is in GByte range, this is a raw data steam.

I was able to decode a noisy color video, so it is normal PAL, FM modulated at 11 MHz.
Next step is to get a better data capture and get a clean software decoding to finally understand all parameters necessary for a usable solution (e.g. is there a deemphasis necessary?).

DaSchaef, 2024-09-15:
I did progress further and did more research, simulation and tries in the past time.
My conclusion is to not proceed on the video side for the moment.
I'll move to audio, but do not expect fast progress, it is in my free time.

The problem with video is to find the best combination of the following requirements, which are hard to combine:
1. Cheap: max. 100€, as otherwise it comes closer to used TKS gateways.
2. Easy: As a hobby project, the people need to assemble themselves and number of units is very tiny.
3. Future proof: Of course there are solutions, but it does not make sense to design one,
where chips, inductors or similar are not available anymore in half a year (fast moving EEE market).

In principle there are two points to solve:
- FM demodulation, where the bandwidth is rather high compared to the modulation frequency
(where requirement 3 is had to reach without FPGA, which violates 1 and 2)
- Digitize the baseband signal without FPGA (which would violate requirement 1 and 2 for hobby users).

In the past there were rather easy solutions, which are still available on the used marked (analog TV tuner chips, cheap PLLs etc), but this would violate 3. These solutions are not produced anymore, since already quite some time.

Going the "direct" computing solution would mean, to e.g. reprogram a suitable SDR device (e.g. I digitized it with the hackRF one clone, ~ 100€). But this again needs another computer, or a custom firmware. 

I think a solution could still be possible, but it is not fun to solve these two main problems with the 3 requirements.

mrtnkhl, 2024-09-16:
@DaSchaef shall we close this issue then or would you like to keep it open?

closed 2024-09-20 by DaSchaef

kripton, 2025-01-03:
Hi, also if this is closed, I would like to give my 2 cents, including some background.
I own a Gira system (no video camera attached), including their TKS-IP-Gateway and some Gira G1 connected via Ethernet. The TKS-IP-Gateway sits on the bus and is able to act as a SIP-server to stream audio AND video to connected clients when someone rings the door.
The flash chip of the gateway is flaky and forgets some bits if not written for some time, resulting in boot failures. For details, see https://www.mikrocontroller.net/topic/531261. Since I was plagued by that, I digged into their bootloader (which changed between firmware v4 and v5 from RedBoot to uBoot and was able to fix my gateway (twice!) and the gateways of two other people who sent me theirs.
However, this is not a good long term solution and I was thinking to replace the gateway's digital/Linux part with something custom and keeping their bus/analog PCB. Thank god I found your project! Perfect substitution for the expensive and buggz TKS-IP-Gateway!

Since I made some pictures of their PCBs, I can share that their TKS-IP-Gateway uses the Sanyo LA9520 chip to demodulate the video. Supports audio as well. Receiver: https://www.alldatasheet.com/datasheet-pdf/pdf/40415/SANYO/LA9520V.html, Transmitter: https://www.alldatasheet.com/datasheet-pdf/pdf/40414/SANYO/LA9511W.html

If we could source that (or a similar chip), video might be a lot easier to do.

Does someone have some spare gdoor v4 adapters or is a group-buy planned the next time? I would definitely be interested, even though I'm more a fan of the RP2040 than the ESP32 ;)

DaSchaef, 2025-01-15:
Thank you! :)

The LA9520 seems to be obsolete or not produced any more.
The video ADC part is an TPS something from Ti, as far as I read on microcontroller.net :)
If really needed, I think I can find the exact part number.

For group buying, please se #19

JustusFluegel, 2026-10-17:
Hi, sorry for bumping this closed issue! I think there may be a viable approach worth exploring, and I'd potentially like to work on some hardware to recover the video signal using a direct-sampling / bandpass-sampling architecture and an inexpensive DSP chip or FPGA (ideally no more than €30). It sounds like an interesting project. :)

IMO, hardware-constrained environments are always fun to experiment with: finding ways to make hardware do things it perhaps wasn't originally designed to do.

I’m currently considering some combination of the dsPIC33, a Lattice ECP5 FPGA (something around ~€25), or the NXP LA9310, with an external ADC for the latter options (e.g. an ADC10040 or similar, at around €20).

In particular, the dsPIC33AK128MC106 might be quite attractive since it can be purchased for less than €3 on DigiKey, even in quantities of one. It also has integrated DSP capabilities and a 40 MS/s ADC, and is an actively supported product, as are the other options mentioned above. The main drawback is its relatively limited processing power, which could make implementing an effective reconstruction algorithm challenging.

> The problem with video is to find the best combination of the following requirements, which are hard to combine:
>
>     1. Cheap: max. 100€, as otherwise it comes closer to used TKS gateways.
>
>     2. [...]

I think a direct-sampling / bandpass-sampling approach could potentially address these concerns, although the assembly aspect might still be somewhat challenging if it requires hand-soldering SMD components rather than having them assembled by a service such as LCSC/JLCPCB.

Another approach I find interesting and worth considering is model-based reconstruction of the signal (see also model-based sparse signal recovery and/or sub-Nyquist sampling), which would exploit known structure in the signal to potentially reduce the required sampling rate.

If this turns out to be feasible, it might be possible to use cheaper DSP chips or microcontrollers with integrated ADCs that would traditionally have far too low a sampling rate to capture the signal directly. Of course, the main challenge would be determining whether the signal has enough exploitable structure and whether the reconstruction can be implemented with the available processing power.

With that in mind, I have a request: I currently don’t have access to an intercom with video (mine is audio-only), so I’d really appreciate it if someone (@DaSchaef, perhaps?) could upload or share a captured signal for me to experiment with, if a recording is still available.

Ideally, I'd be interested in a raw SDR/IQ capture, if available, along with the sampling rate and center frequency. Even a relatively short recording would be a great starting point.

I realize the file might be quite large, but I live near Munich, Germany, and would be happy to send a USB stick or SSD, depending on the file size, along with return postage to a DHL Packstation in Germany if sharing it that way is easier. :)

If anyone happens to live near Munich, Germany, I'd also be happy to meet up and capture the signal together using my Rigol MHO934 oscilloscope (unlocked to 800 MHz analog bandwidth / 4 GSa/s, although I currently only have 500 MHz passive probes, which should still be more than sufficient for this signal).

This could be useful if an existing SDR recording isn't available, or if we'd like to obtain a high-quality capture for experimenting with signal reconstruction. Of course, no worries if that's not convenient! :)

Thanks!