gdoor-org/gdoor #22: DIN Rail Mount

mrtnkhl, 2024-05-17:
To mount the gdoor adapter in an electrical panel next to other Gira intercom components, a 3D printed case that fits onto a DIN rail would be great.

DaSchaef, 2024-05-17:
Here is a 3D render of v3.1
Additionally to the bus connector, either the USB port or 5V pads (shown as holes in the step export) are needed for power.

Height is not 100% sure, as the ESP32mini is plugged into the v3.1 adapter.
[v3.1-step.zip](https://github.com/gdoor-org/gdoor/files/15355771/v3.1-step.zip)

DaSchaef, 2024-10-10:
Please reopen if needed, I currently do not work on it.

Shoomaan, 2024-12-18:
Just a quick design: edited some DIN Rail mount from thingiverse to fit the gDoor Bus Adapter with ESP32 mini on top.
Any ideas for better cable outlet are welcome

[image]
[image]
[gDoor-housing.zip](https://github.com/user-attachments/files/18177081/gDoor-housing.zip)

DaSchaef, 2024-12-28:
Thank you! :)

In future, we can also update the PCB  but currently my time is rather occupied with the audio stuff. We hav a first beta hardware, but there are still bugs to fix first. And it will still take quite some time.

reginastracii, 2026-09-26:
[cable_holes.zip](https://github.com/user-attachments/files/32686411/cable_holes.zip)
Added cable holes in the front and some supports inside. Not ideal but a step...