gdoor-org/gdoor #29: Bus Adapter Procurement Round 3

DaSchaef, 2024-06-19:
As mentioned from @github-k8n  in #19 the connectors seem to be unavailable at JLCPCB.
https://jlcpcb.com/partdetail/Dorabo-DB301V_3_5_2P_GNS/C695629 can be a replacement,
but in the current HW Rev3.1 they would collide. Therefore the connector holes need to be moved a bit to the PCB edge.

I'll hopefully release a "Screw Terminal Version" 3.1-s this evening.
If you want to organize a procurement round, feel free to use this issue here :+1:

DaSchaef, 2024-06-19:
Should work now:
[image]

https://github.com/gdoor-org/gdoor/releases/tag/3.1-1

github-k8n, 2024-06-19:
the 2 unpopulated places in the middle are expected?

DaSchaef, 2024-06-19:
Yes

github-k8n, 2024-06-19:
Ok, ordered 5 , will mention here when they arrive, if they work and then might sell 3-4 for anyone interested (including esp32). Might take a while though (with cheapest delivery option ..)

fabicodes, 2024-06-20:
> Ok, ordered 5 , will mention here when they arrive, if they work and then might sell 3-4 for anyone interested (including esp32). Might take a while though (with cheapest delivery option ..)

I'd be interested in buying one from you then

monnypython, 2024-06-21:
> Ok, ordered 5 , will mention here when they arrive, if they work and then might sell 3-4 for anyone interested (including esp32). Might take a while though (with cheapest delivery option ..)

Hi, I would also like to buy one.

fabermundi, 2024-06-21:
Hello, I would also be interessed.

DaSchaef, 2024-06-24:
For me it looks good, I see the string main+hash in the last images (artifacts).
Thanks alot!

jschroeter, 2024-06-24:
> For me it looks good, I see the string main+hash in the last images (artifacts). Thanks alot!

I guess you wanted to post this in #12

DaSchaef, 2024-06-24:
Ups, yes ^^

rhmunich, 2024-06-30:
> Ok, ordered 5 , will mention here when they arrive, if they work and then might sell 3-4 for anyone interested (including esp32). Might take a while though (with cheapest delivery option ..)

Hello, I am also interested. Would be great if you let me know whether there is still a chance to get one. Thx.

github-k8n, 2024-07-02:
The boards are here and it seems to be working. Unfortunately I did order the wrong ESPs so i don't have those included (just had one of the right ones left over from previous project).
Just the (populated) board for 10 EUR + 5 EUR for packaging/sending  would be my price.
@fabicodes @monnypython @fabermundi , you were the first three and I want to keep two boards so the 3 remaining will go to you.
Not sure how to best proceed here as there is no private chat function.  Maybe if you trust me you can send the 15 EUR WITH your address via paypal (using send to friends to avoid paying extra) to paypal.com@k8n.de and I will send you the board (only the populated board, no esp32 included)
Again, only for the three people referenced above.

fabermundi, 2024-07-03:
> The boards are here and it seems to be working. Unfortunately I did order the wrong ESPs so i don't have those included (just had one of the right ones left over from previous project). Just the (populated) board for 10 EUR + 5 EUR for packaging/sending would be my price. @fabicodes @monnypython @fabermundi , you were the first three and I want to keep two boards so the 3 remaining will go to you. Not sure how to best proceed here as there is no private chat function. Maybe if you trust me you can send the 15 EUR WITH your address via paypal (using send to friends to avoid paying extra) to [paypal.com@k8n.de](mailto:paypal.com@k8n.de) and I will send you the board (only the populated board, no esp32 included) Again, only for the three people referenced above.

Great news. Thank you for that. I've send the money and a mail to the address you sented. Hope you got it.

github-k8n, 2024-07-03:
ok. got the money and the address from all three, will be sending them out today.

For anyone else wanting to start another round, at the moment JLCPCB still has free assembly.

DaSchaef, 2024-07-23:
Is everything working? Can we close the issue?

github-k8n, 2024-07-23:
I guess so, i sent out all three and received a confirmation from one at least and considering i did not hear back from the other two I guess they also arrived.

Now it's just trying to get it to work with esphome which is a completely different issue :D

DaSchaef, 2024-07-23:
in #25 there is some work which we hopefully can integrate, maybe you two can connect?
I'm still working on audio/video. This is a slow process. :)

closed 2024-07-24 by DaSchaef

DaSchaef, 2024-10-09:
Could you please look into #36 and give me feedback about stability (details are in #36) - the more people test, the better.
This would help me to free up some I/Os for audio support. No firmware update needed, only a setting in the web interface of GDoor.