gdoor-org/gdoor #31: looking for bus adaptor

xlemassacre, 2024-08-11:
Hi, I'm very interested in this project. I have been looking for a solution to integrate my Gira system for a long time. As i'm also using esphome in other projects the effort currently done in #25 looks promising!

Does anyone have a bus adapter for sale or is planning a procurement round to order some?

bhuebschen, 2024-08-11:
a friend of mine is interested, but only if he can find enough people

xlemassacre, 2024-08-14:
Ok lets wait for more people joining :) would your friend do the ordering?

ThomasSBW, 2024-08-17:
Dear all,

I was also looking for long time to find a solution for my Gira Audio communication.
When I found this project I was impressed that it should also work with Home Assistatnt.
I order the Adapter by JLCPCB and as everybody knows the limit is 5 Adapter, about this if somebody needs an Adapter last Version let me know.
About the price I am not sure now but I think 8-10Euro plus DHL depense of delivery (Warensendung,Packet, etc.).
The Hardware Version is V3.1-1.

I am also hope that the communication by voice will come soon.
The only problem from my side is that I have no experience in programming. :-(

Thomas

[image]
[image]

DaSchaef, 2024-08-18:
If you need technical support, we are still active - but there is no real news currently therefore it is rather quiet in the repo. I'm working on video and audio (see other issues), but it is a slow progress. (free time, summer etc). The slow progress is tracked in the issues.

bhuebschen, 2024-08-18:
> Dear all,
> 
> I was also looking for long time to find a solution for my Gira Audio communication. When I found this project I was impressed that it should also work with Home Assistatnt. I order the Adapter by JLCPCB and as everybody knows the limit is 5 Adapter, about this if somebody needs an Adapter last Version let me know (2 piece). About the price I am not sure now but I think 8-10Euro plus DHL depense of delivery (Warensendung,Packet, etc.). The Hardware Version is V3.1-1.

I'm interested in one of the adapters :) please let me know how i can send you the delivering details and how i can send you the money

jschroeter, 2024-08-19:
Little suggestion: since GitHub doesn't support private messages, it's probably easiest to open a new private repo and invite the interested people. Then you can share your contact details there.

p08dev, 2024-08-25:
I just ordered 5, if they arrive and work I will post my contact info :)

benoti-git, 2024-09-08:
@p08dev: I am also interested in one adapter, if you still have one spare, I would be happy to buy it. 
Please let me know your contact details or an eBay link.

p08dev, 2024-09-10:
@benoti-git i don't have an ebay link yet, but my email is in my profile :)

p08dev, 2024-09-22:
They are all sold ^^

closed 2024-09-29 by xlemassacre

DaSchaef, 2024-10-09:
Could you please look into #36 and give me feedback about stability (details are in #36) - the more people test, the better.
This would help me to free up some I/Os for audio support. No firmware update needed, only a setting in the web interface of GDoor.