gdoor-org/gdoor #33: Bus Protocol

DaSchaef, 2024-09-20:
I found more infos, which we may be able to use to better understand the protocol,
needs time and starring at raw bus messages to see if it may help us :-)

The Gira Homeserver documentation, seems to give good indications:

**Button:**
1=Light
2=Door Opener
16=Call Request
17=Call Accept

**Source Type (needs translation from German):**
160 = Türstation
161 = Wohnsprechstelle
162 = Wohnsprechstelle Hörer
163 = Bussteuergerät
164 = Schaltaktor
165 = TK-Gateway
166 = Zusatzgong/Busläutewerk
167 = Tasterschnittstelle
168 = TKS-IP Gateway
(e.g. 160 is 0xA0 in hex, this matches what we see on the bus for door station = Türstation)

**Switch Actions:**
Wert 0: Es wird auf AUS-Telegramme gefiltert.
Wert 1: Es wird auf EIN-Telegramme gefiltert.
Wert 2: Es wird auf Toggel-Telegramme gefiltert.

**In multiple pages:**
- Aufsatz (Wertebereich 0 bis 31, indicating 5 bits)
- Tastennummer (Wertebereich 0 bis 127, indicating 7 bit)

**KeyLess-In Gerätetyp:**
(Wertebereich 0 bis 7)
0 = Steuergerät oder KeyLessIn Einheiten
1 = Transponder-Leseeinheit
2 = Fingerprint-Leseeinheit
3 = Codetastatur

**KeyLess Index-Nummer:**
(Wertebereich 0 bis 255)

Beispiel: Finger 17, Transponderkarte 22
Codetastatur/Fingerprint-Leseeinheit
Wert 253: F-Taste gedrückt
Wert 254: Ungültiger Index erkannt
Wert 255: Alte KeyLess-In Firmware

**Nah-/Fernfeld (nur Transponder)**
(Wertebereich 0 bis 1)
0 = Fernfeld
1 = Nahfeld

**TKS-Bus Teilnehmer**
1 = Anrufzustände aller TKS-Communicator beenden
2 = Anrufzustände aller sonstigen TKS-Bus Teilnehmer beenden
3 = Anrufzustände aller TKS-Bus Teilnehmer beenden (auch TKS-Communicator)

**Source:**
http://www.hs-help.net/hshelp/gira/v4_7/de/log16936.html and similar pages

mrtnkhl, 2024-09-21:
Translated text below

**Button:**
1=Light
2=Door Opener
16=Call Request
17=Call Accept

**Source Type**
160 = Door station
161 = Home intercom station
162 = Home intercom station (with receiver)
163 = Bus control unit
164 = Switching actuator
165 = TK gateway
166 = Additional chime / Bus ringer
167 = Push-button interface
168 = TKS-IP gateway

**Switch Actions**
Value 0: It is filtered for OFF telegrams.
Value 1: It is filtered for ON telegrams.
Value 2: It is filtered for Toggle telegrams.

**In multiple pages:**
Attachment (value range 0 to 31, indicating 5 bits)
Button number (value range 0 to 127, indicating 7 bits)

**KeyLess-In device type:**
(Value range 0 to 7)
0 = Control unit or KeyLessIn units
1 = Transponder reader unit
2 = Fingerprint reader unit
3 = Code keypad

**KeyLess index number:**
(Value range 0 to 255)

Example: Finger 17, Transponder card 22
Code keypad/Fingerprint reader unit
Value 253: F-key pressed
Value 254: Invalid index detected
Value 255: Old KeyLess-In firmware

**Near-/Far-field (only transponder)**
(Value range 0 to 1)
0 = Far-field
1 = Near-field

**TKS-Bus participants**
1 = End call states of all TKS-Communicator devices
2 = End call states of all other TKS-Bus participants
3 = End call states of all TKS-Bus participants (including TKS-Communicator devices)

DaSchaef, 2024-09-25:
More findings:

### Fun Fact:
The video indoor station(s) seem to show more menu option, when the whole bus is in programming mode.

### Hardware Types (Indoor, ourdoor etc):
Above values seem to fit. I think we can extend the type list.

### VIDEO_REQUEST
- Param0:
  - Bit0: Turn Light in Outdoor camera off
  - Bit1: Set video frequency in Param1
- Param1: Video Frequency, if Bit1 in Param0 is set

### BUTTON_RING, BUTTON_LIGHT, DOOR_OPEN
- Param0: Most likely button number (and maybe attachment number, see above).
- Param1: 0x60 for normal press, 0xA0 for long press (indicates something in bit 8 and bit 7)

### CALL_INTERNAL
- Param0: Most likely button number (and maybe attachment number, see above).
- Param1: always 0x60 for normal press?

### Button
- Param0: Always bit 7 seems to be set (0x40), lower bits seem to indicate button number (1,2,3...)
- Param1: 0x40 for ON, 0x50 for OFF (currently hard to detect which bits are really used)

closed 2024-10-10 by DaSchaef