gdoor-org/gdoor #35: Bus Adapter Procurement - October Edition

DennisGeisse, 2024-10-01:
Hi, 

I just recently discovered this project and would like to get involved, for example integration in ESPHome (discussed in #25) is something I might be able to contribute to.

However, first things first: I don't have a bus adapter yet. Is there anybody that has a spare one that they would like to sell? 

Looking forward to join this community :)

DaSchaef, 2024-10-03:
For ESP Home you may look at #25 and contact the people, I'm interested to include also ESPhome into GDoor, but I'm no HA user (I'm currently working on further understanding the bus messages and in future I want to look into audio).

DaSchaef, 2024-10-24:
If you did not yet procure a board you may want to look at  #38 :)

closed 2024-10-30 by DennisGeisse