gdoor-org/gdoor #38: Adapters for Sell

marco2901, 2024-10-23:
Hi, 

I currently have 4 3.1-1 Boards for sell. Each 12€ plus shipping.

BR
Marco

dschlegel, 2024-10-27:
Hi Marco,
where are you shipping from?
Regards
Daniel

marco2901, 2024-10-27:
Hi @dschlegel,

from Germany. So I could send it as "Warensendung" to you.

BR
Marco

dschlegel, 2024-10-27:
Sounds good. Can you email me at Daniel.schlegel@posteo.net? Then we can discuss the details.

devcomputer, 2024-10-28:
i am very much interested as my landlord just installed brand new 1250, i have HA running and some m5stack EPSs here.
but i do not understand how to connect the gdoor thing to the ap1250?
documentation says:"After successfully setting up the adapter, you can connect it to the Gira TKS Bus."
how? can i just wire it to the two bus cables that are easily accessible or do I need to solder? i dont want to break it and anger the landlord.

DaSchaef, 2024-10-28:
Yes you connect directly to the bus cables and not to the indoor station.
You can connect anywhere, where you have access to the bus cable.
No soldering required on Gdoor side.

Remember, soldering is sadly required for the ESP board which is plugged into the GDoor adapter, as the pin headers do not come pre-soldered (https://www.az-delivery.de/en/products/esp32-d1-mini).

This is the bus connector:
[image]

Please note, this is an image of v3.1 with a plugable connector,
the new v3.1-1 has only a screw terminal (because of parts availability).

Please also see the warning at the end of the Getting Started page:
https://gdoor-org.github.io/documentation/getting-started.html#h-warning

devcomputer, 2024-10-30:
pretty cool. do i have to use a esp32-d1-mini? i have a box full of AtomS3 Lite ESP32S3 Dev Kits here.
so i would like to buy a board anyway. @marco2901 where is your contactinfo?

DaSchaef, 2024-10-30:
> pretty cool. do i have to use a esp32-d1-mini? i have a box full of AtomS3 Lite ESP32S3 Dev Kits here. so i would like to buy a board anyway.
You may, depending on which I/Os are free, but it will be "ugly" (see my development setup ;-) https://github.com/gdoor-org/gdoor/issues/20#issuecomment-2435736075)

I see multiple options:
- use EspMini, plug and play
- use different ESP32 board (no ESP32Cxyz, only ESP32), connect I/Os via jumper cables.
- use different ESP32 board, recompile firmware to match your available I/O

I recommend ESP32Mini ;-)

marco2901, 2024-10-30:
> pretty cool. do i have to use a esp32-d1-mini? i have a box full of AtomS3 Lite ESP32S3 Dev Kits here. so i would like to buy a board anyway. @marco2901 where is your contactinfo?

Hi, please just let me your mail address here. I will send you the details.

BR
Marco

TilldaKiNG, 2024-10-30:
> > pretty cool. do i have to use a esp32-d1-mini? i have a box full of AtomS3 Lite ESP32S3 Dev Kits here. so i would like to buy a board anyway. @marco2901 where is your contactinfo?
> 
> Hi, please just let me your mail address here. I will send you the details.
> 
> BR Marco

Hi Marco,
I got a new Gira TKS installed today and found this great project. Since i had my old doorbell in HS. I would like the same for the new one. I am from Germany as well an would like to buy a GDoor PCD, if you have stock. You can contact me via huepfen-08-brote@icloud.com thanks!

rstein, 2024-11-08:
Hi @marco2901, I‘m interested in buying a board from you. If you‘ve got any left, please get in touch! 😊 
rs_any@icloud.com

Secarius, 2024-11-16:
@marco2901 Do you still have one or two left?

marco2901, 2024-11-16:
Hi @Secarius ,

yes, 2, please send me your contactcdetails (mail).

BR

Secarius, 2024-11-16:
> Hi @Secarius ,
> 
> yes, 2, please send me your contactcdetails (mail).
> 
> BR

Secarius12345@googlemail.com

allgrinder, 2024-11-18:
Hey Marco, I would appreciate an email – shipping would be to Germany. PayPal is welcome.
black-fighters@web.de

Shoomaan, 2024-11-22:
@marco2901 
Hi Marco, do you still have one Board left? 
If yes, I would be interested in buying one.

devcomputer, 2024-11-22:
oh btw did someone already create a cool case for it? didnt find anything on thingivese.

mrtnkhl, 2024-11-23:
@devcomputer check out https://github.com/gdoor-org/gdoor/issues/22

so far no case has been made but a 3D printed DIN rail case like you can get for the Shelly devices would be amazing. @DaSchaef has linked some step files in the issue, maybe that helps!

lucasheld, 2024-11-25:
@Shoomaan 
I have one board left. What's your mail address?

c4ve-d, 2024-12-04:
Hi

I‘m interested in buying a board. Someone got any left?
Since I'm in Germany preferred from Germany (or EU).
Please get in touch!

Thank you in advance

lucasheld, 2024-12-04:
@c4ve-d I'm from Germany and have one board left. What's your email address?

c4ve-d, 2024-12-04:
@lucasheld c4ve@gmx.de

ix2, 2024-12-06:
Hi, I am interested in buying a board. Anybody have one available? BTW, my location ist Germany (Ruhrgebiet).

DaSchaef, 2024-12-17:
I'm closing this one in favor of #19. @ix2 If you still need one, you may look into #19 it seems thre are adapters available.

closed 2024-12-17 by DaSchaef