gdoor-org/gdoor #39: Extend support to Urmet 2Voice BUS - if possible

stich86, 2024-11-12:
hi @DaSchaef,

I've found your amazing work searching for someone that has tried to reverse Urmet 2Voice BUS.
My intercom is using this system (Urmet is an Italian company that is producing intercom and some other audio\video stuff).

The system is based on 2-wire, non polarized, that runs from 48V to 50V (for very log distance I presume). Each internal unit, without video, is using an STM32G02 as MCU to parse BUS data. There are a row of dip-switches to select the ID and connect up to 4 intercom inside a single apartment. 
Audio and video seems analogs (like Comelit SimpleBus), but not 100% sure. 

Now my first concern is how to attach the BUS and start reversing it using an oscilloscope. 

Do you have any hint?

Thanks in advance for any support!

DaSchaef, 2024-11-12:
I'm aware of multiple modulation schemes, most (SIEDLE, COMELIT) are based on voltage levels, only Gira so far used a "real/RF" modulation like scheme.

For Audio/video you could risk to just connect a cheap USB audio dongle and a raspberry and try it out.

For digital messaging (door opening etc):
I doubt that the GDoor adapter would be able to decode another bus type, it is rather specialized to the Gira modulation.
Nevertheless, I see the possibility, that it may be similar to comelit, where there are also open source projects here on Github.
So you may ask there.

If you want to do it yourself with the oscilloscope it would first be very helpful what exact scope do you have?
Most good scopes have rather high input voltage ratings (100V and more), but cheap scopes may be sensitive.

So:
- What scope do you have?
- What is your level of experience with scopes?

The principle way to do it:
- Find the "Ground" wire, connect the scope ground clip to it.
- Set your scope to "AC" (which often reduces risk of damage)
- Set your scope to 5V/div
- Connect the scope to the signal wire
- Set your Trigger to "Auto", so that you will always acquire data
- Set Timescope to "rather long" while still having a "fast" updated display (so that half a second fits on the screen, roughly)
- A key problem with cheap scope is the dead time: After acquiring data, the scope needs some milli-seconds (sometimes seconds!!!) until it can acquire new data. So you will have "gaps" and if your interesting data is in a gap, you will not see it.
- The trick ist to generate a lot of data, so that the likelihood to fall into a acquisition gap is very low :)
- Generate Bus messages, look at the scope to get a feeling: Do you see changes in the lines? How high are they? how long do they take?
- Reset your settings: Narrow down vertical (voltage) and horizontal (time)
- Generate Bus messages again, iterate until you have a nice setting
- Now set your Trigger to manual. This means acquisition will start only at a certain event (voltage ramp), select a fitting voltage level and rising or falling ramp (slope).
- Generate a Bus message, this message should now stay on your scope screen. So you can look at it more closely.
- Sometimes you may want the trigger to "Single" instead of normal. So it triggers only once,
any you manually need to re-arm (start) the triggering afterwards.
- Now you can iterate to find settings to capture a nice looking, informative image: Full message, zoomed parts of the message, etc.
Important part is that the signal uses as much "space" on your screen as possible (vertical and horizontal), no "unused" space.

1. If you can share these pictures, we can further see what to do with it.
2. Often you can also export the raw data shown on the screen via USB/Lan: Sometimes this is helpful,
as one can "zoom" into it a full bus message later on etc. (Depends on scope)

But this is a project in its own, I have no Urmet parts and I'm currently working on the Audio support here for the Gira bus.
I can only partly support you.

DaSchaef, 2024-11-22:
Please reopen as needed, when you have more information / images.

stich86, 2024-12-01:
Hello,

I've played today with the equipment (found some stuff to create a small lab). Here is an exemple of push "Open door" button.

Any suggestion?

[image]

DaSchaef, 2024-12-01:
You need more pictures, zoom in, zoomed out.
Such a single picture can not transfer all information. For the full communication it is too short, for the modulation scheme it is too long.

What I find strange is that it looks like there is always an idle oscillation, which would be rather unusual.

stich86, 2024-12-01:
Tomorrow I'll try to do a live capture on an usb stick, may be in audio way so can be analyzed by Audiacity

Yes you are right.. there is some sort of "static" noise on the background. I don't know if it's related to to the cabling or something else..

closed 2024-12-28 by DaSchaef