gdoor-org/gdoor #41: Interference on Indoor Station Video

allgrinder, 2024-11-23:
Hier ist die Übersetzung ins Englische:

Hi,

I received my board yesterday and immediately connected it to the bus. It was a quick and dirty setup, but I am receiving messages from the bus. Some messages might be duplicated, but overall, it works so far.

Now to my problem: As long as the ESP32 and the board are powered on, the video image on my indoor station has interference. Once the ESP is powered off, the image is perfect again. What could be causing this?

Best regards,

DaSchaef, 2024-11-23:
Please see #40 I think it is the same issue: Ground loop (Brummschleife).
Test it with USB power bank to ensure that this is the case.
If yes, you can correct via different USB power supply, electrical phase, install GDoor closer to the video controller and/or a self build filter as shown in the issue.

allgrinder, 2024-11-23:
Yes, indeed, everything works perfectly with a power bank! Thank you.

Do you have a link for a suitable power adapter?

DaSchaef, 2024-11-23:
Sadly no, it may also really depend on the electrical installation. A generic solution is filtering the power supply,
via Common Mode Choke / Balun / "Gleichtakt-Drossel" (which Gira also builds into its devices,
but we can not implement it, because of parts availability at JLCPCB).
Luckily these filters can be implemented DIY without soldering etc (see #40 for examples).
Some power supplies also implement these filters.

I would suggest:
First try different supplies you already have.
If possible also try different power sockets / locations (phases).
If nothing helps, DIY a filter, you can either filter at GDoor bus connection and/or USB supply,
you need to try.

I would love if you can give feedback, so we can document it - I "sadly" have no problems on my bench setup for development. And in case nothing works I can support further, check the filter etc.

DaSchaef, 2024-11-23:
Looks like a good supply, worth a try - but still I can not say for sure that it will work.

DaSchaef, 2024-12-03:
Did you have a chance to check already? Can we close it?

allgrinder, 2024-12-07:
Not done yet…

allgrinder, 2024-12-13:
With the power supply linked above, I have no issues. However, I also haven’t connected a ground

DaSchaef, 2024-12-17:
Ok and thanks for the feedback, I'll close. If there are problem, just reopen :)