gdoor-org/gdoor #43: Home Assistant (Yellow) Serial Command Issues / Config Update

mrtnkhl, 2024-12-03:
https://gdoor-org.github.io/documentation/homeassistant.html

**Sensor configuration documentation**

Correction for typo needed, documentation currently reading:
```
sensors:
   - platform: serial
     name: "GDoor"
     serial_port: /dev/ttyUSB0 # change device if needed
     baudrate: 115200
```

Should read  (sensor instead of sensors):
```
sensor:
   - platform: serial
     name: "GDoor"
     serial_port: /dev/ttyUSB0 # change device if needed
     baudrate: 115200
```

**Sending command via serial**
Does not work on Home Assistant Yellow (HAOS). Sensor works when config corrected as per above, but sending command via serial not possible. Running latest audio beta firmware.

configuration.yaml
```
shell_command:
  gdoor_open_street: echo -e '020031ECF3EB0060A103C27A3D' > /dev/ttyACM0
  gdoor_open_courtyard: echo -e '020031ECF3EB0060A11A2F76BD' > /dev/ttyACM0
```

Error in log
```
Logger: homeassistant.components.shell_command
Source: components/shell_command/__init__.py:95
integration: Shell Command (documentation, issues)
First occurred: 11:16:44 (5 occurrences)
Last logged: 11:21:29

Timed out running command: `echo -e '020031ECF3EB0060A103C27A3D' > /dev/ttyACM0`, after: 60s
Timed out running command: `echo -e '020031ECF3EB0060A11A2F76BD' > /dev/ttyACM0`, after: 60s
```

jschroeter, 2024-12-27:
Thanks, I fixed the typo in https://github.com/gdoor-org/gdoor-org.github.io/pull/4

I guess the not working sending over serial is just related to the WIP audio firmware, @DaSchaef?

DaSchaef, 2024-12-28:
Correct, the audio firmware can not send as documented currently.