gdoor-org/gdoor #45: Debug logging / console via HTTP interface

mrtnkhl, 2024-12-03:
For locations that do not allow for serial access to the board, it would be ideal if logs / console would be accessible via the gdoor firmware http interface

DaSchaef, 2024-12-03:
I can for sure say that a debug logging over WiFi will not be possible. Only higher level logging could be done, but if it is higher level the benefit and gained information of it may be questionable.

This is a technical limitation of small embedded systems:
- Not much flash storage
- Writing flash storage decreases lifetime (systems are optimized for write rare, read often)
- Sending out via WiFi already means that a lot subsystems if not all need to work. E.g. how to send out debug messages for wifi, over wifi, if wifi makes problems ;-)

mrtnkhl, 2024-12-04:
Higher level would be great, just to verify that the gdoor device is happy and working or to manually send commands, e.g. send a payload directly from the firmware and not via MQTT or serial. Stream into the browser window would be sufficient, no need to keep log in flash.

Ideally it would be possible to follow along what the unit is doing high level and issue commands so that correct function can be verified.

DaSchaef, 2024-12-04:
This may be a feature more in the future, as it would also mean to change the WiFiManager