gdoor-org/gdoor #47: Ethernet Interface

mrtnkhl, 2024-12-04:
Often times wiring for the Gira Bus is installed / terminated in metal cabinets (Unterverteilung) which prevent WIFI signals from reaching access points.

Space permitting, it might be helpful to have an Ethernet interface with an RJ45 connector on the gdoor board. That way, an Ethernet cable can be plugged in and no wifi needs to be used. An ESP32 board with an integrated Ethernet interface could be an alternative.

The cherry on the cake would be POE so that only one Ethernet cable is sufficient to power the gdoor assembly. Galvanic isolation and available ports at the board manufacturer could be an issue here.

DaSchaef, 2024-12-04:
If we find someone we may be able to do it, but because of the EMI aspects, this is currently out of scope. Too risky to create distortion of audio/video on the bus.
Main issue is, that we are limited to what EEE parts are available at JLCPCB and I try to select long term available parts,
so that GDoor hardware does not constantly need to be modified to match parts availability.
The audio CODEC is already rather borderline and I assume will create availability issues in the future.

closed 2024-12-04 by DaSchaef

mrtnkhl, 2024-12-04:
POE support would be a nice to have but not critical since a POE to RJ45 and USB C splitter could be used or the ESP32 board be fed via a USB C power supply. If the risk for basic Ethernet to create a ground loop is too high, that's a different story.

Possible Ethernet chipsets could be:

- LAN8720 (RMII)
- RTL8201 (RMII)
- DP83848 (RMII)
- IP101 (RMII)
- JL1101 (RMII)
- KSZ8081 (RMII)
- KSZ8081RNA (RMII)
- W5500 (SPI)

Those seem to be pretty common, especially the W5500 is available in the tens of thousands.

kripton, 2025-01-03:
I do have limited experience with the ENC28J60 from Microchip and more experience with the W5500 from WizNet (both connected via SPI), however on the RP2040, not on the ESP32.
There are also lots of boards combining the two chips, also with the ESP32, of course.
What you need to keep in mind: At least the W5500 comes without a MAC address. So for hobbyists, either make up some random MAC (as virtual machines do), disable WiFi and use its MAC address instead of buy an I2C EEPROM with a MAC address stored on it (Microchip sells them, for example: https://www.microchip.com/en-us/parametric-search/535)

c4ve-d, 2025-01-20:
Hello

I have already dealt with exactly the same topic. The issue of separating the potential can be easily solved with the right hardware: ESP32-POE-ISO by OLIMEX 
More info about the board here --> https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

It separates the electrical potential between the network and the ESP chip and uses the supported network controller LAN8710A (LAN8720 compatible) and runs on PoE (or USB power supply).

I just had to include LAN support and change the pins for IO.
The only hardware limitation I have found so far is that on this board the analog outputs are already used by the network chip and therefore the adjustable output has to be omitted. 
And yes, it is connected via wires and does not fit natively on the IO pins.

I've already gotten the software to run on the LAN with small adjustments, but I'm faced with the problem that the GDOOR routines only run when the WLAN is connected. If the wifi is not connected but the wired network is connected, the actual GDOOR application does not run. As far as I understand it, it has a logic that checks the wifi connection and the actual software only runs when wifi is active and connected.

Unfortunately, I'm not a software developer and I'm not making much progress in adapting the software. Can someone maybe support me with the needed adjustments?

DaSchaef, 2025-01-20:
Hey, that sounds very good!

Adapting it "cleanly" is a bit of work, because of the WiFiManager library we use :-/

Are you able to compile the software yourself? My idea would be a rather "hacky / hard coded" variant for your needs.
You would loose the web config page

c4ve-d, 2025-01-20:
Hi

Yes, I downloaded the Git repository and then adjusted the configuration (ESP32 library, IO pins, Ethernet).
Then I compiled the software and transferred it to the ESP.

As I said, I'm not a software developer, so nothing pretty but rather quick and dirty with the help of google and trial and error... :-)

I was able to connect the ESP to Ehernet and an IP was obtained via DHCP. 

When connected via WiFi, the ESP32-POE-ISO runs and receives data from the Gira bus.

I failed so far with the configuration or disable or whatever of the WifiManager.

DaSchaef, 2025-01-20:
Did you change the code to get the wired ethernet working? I read it that way.

In that case:

1.

https://github.com/gdoor-org/gdoor/blob/main/firmware/esp32/gdoor/main.cpp

Disable all "WIFI_HELPER" related stuff, in case variables are read, just use hard coded content (most settings examples can be seen in https://github.com/gdoor-org/gdoor/blob/main/firmware/esp32/gdoor/src/defines.h)

2.

Now we need to convince MQTT that you have a valid connection:
https://github.com/gdoor-org/gdoor/blob/main/firmware/esp32/gdoor/src/mqtt_helper.cpp

Disable the WiFi connection state check:
https://github.com/gdoor-org/gdoor/blob/feb611d1c08d6f1e54a5dbc9a58bdd9572465d75/firmware/esp32/gdoor/src/mqtt_helper.cpp#L210

(remove the if case)

You may want to change the code to monitor the wired connection, so that MQTT reconnects, if the wired connection is lost and established later on again.

Hope this helps you 👍