gdoor-org/gdoor #49: Update time via NTP

mrtnkhl, 2024-12-04:
Ideally the firmware would support pulling the current time via NTP so that log messages (e.g. in debug) show the right time.

Currently there is no way to set time.

Time could be set via the HTTP webinterface, ideally a primary and a secondary time server could be set as a URI that can get resolved or as a plain IP address. URI preferred for DNS updates.

DaSchaef, 2024-12-04:
I do not want to make the firmware more complicated. It is a embedded small device, no computer. :)
Time tagging can be done on receiving (local / PC) end, it needs to be done nevertheless via USB - because of the flash/storage limitations.

Contributions welcome, reopen as needed :)

closed 2024-12-04 by DaSchaef

mrtnkhl, 2024-12-04:
Main motivation was that the SIP stack seems to use time. Not sure if that is relevant for SIP sessions and might be verified by the registrar/proxy

```
I (18202795) SIP: [1970-01-01/05:03:10]=======================>>
I (18202980) SIP: [1970-01-01/05:03:10]<<=====READ 0656 bytes==
I (18202980) SIP: 

```

DaSchaef, 2024-12-04:
Good point! Normally it should work - if we see problems based on time, I propose to re-open when we see that "time makes problem" :+1:

mrtnkhl, 2024-12-04:
Back to the 70s for now haha