gdoor-org/gdoor #51: No BusID via Hassio

allgrinder, 2024-12-15:
Hey,
unfortunately, I can't retrieve the BUS-ID as described in the documentation, so I can't control the door via Home Assistant. What am I doing wrong?

Best regards,

[image]

mrtnkhl, 2024-12-17:
Hi @allgrinder,

Can you give a bit more details on what you did step by step? Did you see any BUS messages?

allgrinder, 2024-12-17:
Ah sorry, that was my mistake.
In the end, I simply activated and tested the video function or door opening on the indoor unit. Gdoor was integrated via MQTT...
I also received a corresponding entry in the logbook. Some of them are duplicated, but that’s not a big issue for now... I’ll look into that next.
According to the documentation: https://gdoor-org.github.io/documentation/homeassistant.html, the result in the logbook should look like this:

GDoor changed to {"action": "BUTTON_RING", "parameters": "0360", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "011011A286FD0360A04A"}

Apparently, the parameter is crucial... but I’m not receiving any parameter.

Also, the examples for typical use cases are not working either... I keep getting the message "shell command not found," which is surprising to me.

DaSchaef, 2024-12-17:
I can not comment on HA (I'm a OpenHab user ^^).

> Apparently, the parameter is crucial... but I’m not receiving any parameter.

You received them :-)

> {"action": "BUTTON_RING", "parameters": "0360", "source": "A286FD", "destination": "000000", "type": "OUTDOOR", "busdata": "011011A286FD0360A04A"}

"parameters" =  "0360"

The parameters field has a different meaning for different bus actions.
For ring button pressed ("BUTTON_RING"), it mostly codes which button on the outdoor station was pressed, if there are e.g. multiple stations in the house. Therefore it is important for you, to use the value as filter, to not react to all buttons :)

DaSchaef, 2024-12-17:
If you do not receive anything, you may want to try http://mqtt-explorer.com/ to "listen" to the GDoor ---> HomeAssistent communication. So we can avoid that HA is hiding the GDoor message from you somehow.

DaSchaef, 2024-12-17:
This may indicate that you do not receive any events? The "IDLE" is from GDoor itself and only a kind of keepalive ("I'm here") packet.

You should receive events, when you do something with the bus e.g. press the door button on the outdoor station. For testing, you can also use the light button on the indoor station. This is often easier to reach ;)

If you do not receive any bus message, we need to investigate further:
- Is GDoor connected to the bus?
- Can you listen to the raw MQTT messages (e.g. http://mqtt-explorer.com/) so that we can take HomeAssistant out of the possible error sources
- You should see MQTT messages on the topic, also set up in HA in the MQTT Explorer.

allgrinder, 2024-12-17:
Yes, I do get all the events displayed correctly. But they just come without parameters.
[image]

allgrinder, 2024-12-17:
All good! Thanks for your support. My post with the screenshot at 17:14 shows the details of a Button Ring. Once I’m home and have some time, I’ll install the Explorer.

DaSchaef, 2024-12-17:
The screenshot at 17:14 is an IDLE message, no BUTTON_RING ? This is why the parameter is empty in this mentioned image :)

allgrinder, 2024-12-17:
Yes, that confuses me as well… I click on the corresponding event, and it only shows Idle.

allgrinder, 2024-12-17:
It's really strange – MQTT Explorer shows the parameter, and Home Assistant also displays the various statuses. But the bus adapter's log continuously shows 'Idle' or doesn't indicate that another status briefly appeared.

[image]

[image]

allgrinder, 2024-12-18:
The ticket can be closed. I can control the door via MQTT and receive a notification when the doorbell rings. That was my goal, and it works! Thank you for your help!

jschroeter, 2024-12-18:
Did you check the [Logbook](https://www.home-assistant.io/integrations/logbook/) in HA? Its enabled by default and usually doesn’t need to be configured. There you should see all events including details.
In the History view you’ll mostly only see the details of the last event, which is always BUS_IDLE.

jschroeter, 2024-12-28:
Reopening this as we indeed have 2 problems here - I didn't notice them so far as I was mainly using the serial integration:

1. due to [this line](https://github.com/gdoor-org/gdoor/blob/feb611d1c08d6f1e54a5dbc9a58bdd9572465d75/firmware/esp32/gdoor/src/mqtt_helper.cpp#L132), HA only shows the `action` in the Logbook. Unfortunately HA only allows to see the JSON attributes for the latest state, which is always `BUS_IDLE` (as we send this action after every other action) and therefore you can't see e.g. the `parameters`.
  [image]

2. quite frequently HA reports that the sensor is unavailable/unknown and actions are missing on the HA sensor (but they do appear when listening to the MQTT topic directly). I'm pretty sure this happens because HA only allows 255 characters for a value and we often exceed this.

Suggestions:
- to fix 1),
  - A) we change mqtt_helper.cpp#L132 to `"value_template": "{{ value }}"` which would display the whole JSON as string. Works but looks a bit clunky, especially in the history view.
  [image]
  [image]

  - B) or to `"value_template": "{{ value_json.action }} {{ value_json.source }} {{ value_json.destination }} {{ value_json.parameters }}"`. Looks cleaner but needs explanation what these values actually are. Nice side effect: since the event id is not included, equal actions are shown with the same color code.
  [image]
  [image]

  - C) we drop the BUS_IDLE action again, I personally still don't like it and so far don't have a use case for it [as discussed before](https://github.com/gdoor-org/gdoor/issues/27), but I think some find it useful.

- to fix 2)
  - D) we remove the `raw` attribute as it duplicates `busdata` anyway. By that we should stay under 255 chars.
  - E) we move it and maybe other debug-related attributes (`valid`, newly added `busvoltage`) to an own MQTT topic, e.g. `gdoor/debug` or `gdoor/rx_debug`

Any opinions? I'd vote for C) and/or B) + E).

allgrinder, 2024-12-29:
Yes, @jschroeter is right... it is indeed a bit confusing. But once you're aware of the peculiarity, you can deal with it. I can control the door, receive signals, and perform actions... so I'm very satisfied :)

But C is probably pragmatic.