gdoor-org/gdoor #52: MQTT Messages lost after restart

Shoomaan, 2024-12-18:
I've added some code to my configuration.yaml in homeassistant to create a sensor for the doorbell and also a button / lock for the opener. After restart the devices get unavailable until the Bus Sensor changes it's state.
Perhaps we could fix this by retaining the messages?

here is my config for homeassistant, perhaps you want to add it to the homeassistant section or even better the firmware creates the devices with ha-discovery.

```
button:
  - name: Türöffner
    unique_id: gDoor_opener_button
    command_topic: "gdoor/bus_tx"
    payload_press: "0200311C53940000A2E882F638"
    enabled_by_default: true
    qos: 0
    retain: false
    availability_topic: "homeassistant/sensor/gdoor/data/config/78_EE_4C_01_68_C8"

    device:
     {
          sw_version: dev,
          model: ESP32 (78:EE:4C:01:68:C8),
          configuration_url: http://192.168.178.179,
          identifiers: gdoor_78_EE_4C_01_68_C8,
          name: GDoor Adapter,
      }

      

      }
      
      
lock:
  - name: Türöffner
    unique_id: gDoor_opener_lock
    command_topic: "gdoor/bus_tx"
    payload_unlock: "0200311C53940000A2E882F638"
    payload_lock: ""
    enabled_by_default: true
    qos: 0
    retain: false
    availability_topic: "homeassistant/sensor/gdoor/data/config/78_EE_4C_01_68_C8"
    state_topic: "gdoor/bus_rx"
    json_attributes_topic: gdoor/bus_rx
    value_template: '{{ value_json.action }}'
    state_locked: "BUS_IDLE"
    state_unlocked: "DOOR_OPEN"
    device:
      {
          sw_version: dev,
          model: ESP32 (78:EE:4C:01:68:C8),
          configuration_url: http://192.168.178.179,
          identifiers: gdoor_78_EE_4C_01_68_C8,
          name: GDoor Adapter,
      }
      
binary_sensor:
  - unique_id: klingel_eg
    name: "Klingel EG"
    state_topic: "gdoor/bus_rx"
    value_template: >
      {% if value_json.action == "BUTTON_RING" and value_json.parameters == "0360" %}
        true
      {% elif value_json.action == "BUS_IDLE" %}
        false
      {% else %}
        false
      {% endif %}
    payload_on: "true"
    payload_off: "false"
    device_class: occupancy
    icon: mdi:bell
    availability_topic: "homeassistant/sensor/gdoor/data/config/78_EE_4C_01_68_C8"
    device:
      manufacturer: GDoor Project
      sw_version: dev
      model: ESP32 (78:EE:4C:01:68:C8)
      configuration_url: http://192.168.178.179
      identifiers: gdoor_78_EE_4C_01_68_C8
      name: GDoor Adapter
```

jschroeter, 2024-12-27:
> Perhaps we could fix this by retaining the messages?

You mean by setting the `retain` flag of the MQTT messages sent by the adapter?

> perhaps you want to add it to the homeassistant section or even better the firmware creates the devices with ha-discovery

I just improved the HA docs a bit with https://github.com/gdoor-org/gdoor-org.github.io/pull/4, but not really covering what you suggested.
I like your ideas, we had a similar discussion before: https://github.com/gdoor-org/gdoor/issues/26
Maybe a simple (but not perfect) solution would be to just publish a HA auto-discovery device config for a few typical actions like BUTTON_RING and OPEN_DOOR. I think for the buttons this could be straight-forward, so HA would create a new sensor when a button is pressed for the first time. This may also work for the door locks?!

DaSchaef, 2024-12-28:
What es meant here, what restart? HA? If HA restart, we may need to regulary send the discovery message, or use the IDLE message as availability topic.